Xinyuan-LilyGO/lilygo-micropython #2: Got it displaying text on the screen but where's the rest of the functionality?

scumola, 2022-02-23:
Looks like the micropython port was just a proof-of-concept.  I can clear the screen, display a JPG and draw some text, but that's it.  No "draw a dot/line/circle/arc" functions, no "change the font" function.  Nothing too fancy.  Only the basics have been implemented it seems. Also, I'm seeing some ghosting but it's intermittent.  Sometimes the screen has ghosting, sometimes it doesn't
[image]
[image]
.

liangyingy, 2022-02-24:
1. Regarding "draw a dot/line/circle/arc" functions, use framebuffer or bytearray to draw, and then call EPD47.bitmap() to display. It is completely possible to use python.

2. We have also discovered the problem of ghosting, and we have not found the reason for it.

3. touchscreen can refer to https://github.com/Xinyuan-LilyGO/lilygo-micropython/blob/master/extmod/touch/l58.py

scumola, 2022-02-28:
So how do I use the touchpad?

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.17.0', version='65de710 on 2022-02-14', machine='ESP32 LILYGO EPD47 with ESP32')
>>> import epd
>>> dir(epd)
['__class__', '__name__', 'EPD47']
>>> import touch
>>> dir(touch)
['__class__', '__name__', '__file__']
```

scumola, 2022-02-28:
Got the touchscreen working!

```
from l58 import L58
from time import sleep
from machine import I2C,Pin

tp = L58(I2C(1, scl=Pin(14), sda=Pin(15), freq=400000), 0x5A)
tp.begin()
while True:
    for i in range(0, tp.scanPoint()):
        data = tp.getPoint()
        print(data)
    sleep(0.1)
```

closed 2022-07-06 by liangyingy