Xinyuan-LilyGO/lilygo-micropython #3: SD card not recognized

scumola, 2022-02-28:
I bought the SD card adapter and inserted a brand new FAT-formatted 32GB SD card, but when I try to access the card, I get:

>>> import machine
>>> sd = machine.SDCard()
  Traceback (most recent call last):
    File <stdin>, line 1, in <module>
  OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>>

scumola, 2022-02-28:
This is the official lillygo adapter I'm using

[image]

liangyingy, 2022-03-01:
I don't have a 32GB SD card on hand, the following code works fine on a 16GB SD card:

```python
import machine
import os

sd = machine.SDCard(slot = 2, width = 1, sck = 14, miso = 12, mosi = 13, cs = 15)

sd.info()
os.mount(sd, "/sd")
os.listdir('/sd')
```

scumola, 2022-03-01:
I added SKIPSD to / so micropython doesn't try to boot off the SD card and it works!

Bug: any kind of guru meditation error (like the framebuffer crashes) kills micropython and makes it unresponsive even with multiple power cycles until I physically unplug the SD card adapter and reboot the board.

closed 2022-03-01 by scumola

He-Peter-Huang, 2023-12-16:
I'm using the LilyGo T-RGB, I'm having trouble setting up the SDCard, the CS pin is connected to P07 on the XL9535. What should I put in for the SDCard cs parameter?