Xinyuan-LilyGO/lilygo-micropython #4: Framebuffer example crashes micropython

scumola, 2022-02-28:
Getting a reboot when trying to run this:
```
import epd
import framebuf
epd47 = epd.EPD47()
epd47.power(True)
epd47.clear()
fbuf = framebuf.FrameBuffer(bytearray(100*10*2), 100, 10, framebuf.GS4_HMSB)
fbuf.fill(0)
fbuf.text('MicroPython!', 0, 0, 0xffff)
fbuf.hline(0, 9, 96, 0xffff)
epd47.bitmap(fbuf,0,0,100,100)
```

Output:
```
>>> %Run -c $EDITOR_CONTENT
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d2953  PS      : 0x00060d30  A0      : 0x800e3bdc  A1      : 0x3ffcf1d0  
A2      : 0x000a0064  A3      : 0x000a0064  A4      : 0x3ffcf1ec  A5      : 0x00000004  
A6      : 0x00000060  A7      : 0x3ffcf1d0  A8      : 0x800d2940  A9      : 0x3ffcf1b0  
A10     : 0x000a0064  A11     : 0x3ffcf1e8  A12     : 0x3ffcf1ec  A13     : 0x00000006  
A14     : 0x0000000a  A15     : 0x0000ffff  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000a0064  LBEG    : 0x4009320e  LEND    : 0x40093219  LCOUNT  : 0x00000000  

ELF file SHA256: 59e057f1656a3cc9

Backtrace: 0x400d2950:0x3ffcf1d0 0x400e3bd9:0x3ffcf220 0x400e9a69:0x3ffcf250 0x400e9bba:0x3ffcf270 0x400ec245:0x3ffcf290 0x400e3c64:0x3ffcf330 0x400e9a69:0x3ffcf390 0x400e9a92:0x3ffcf3b0 0x40117a3e:0x3ffcf3d0 0x40117cd5:0x3ffcf460 0x400fbb55:0x3ffcf4b0

Rebooting...
ets Jul 29 2019 12:21:46
```

liangyingy, 2022-03-01:
The following is the correct use of epd47.bitmap(), it will not restart.

[LilyGo-EPD47 using in micropython](https://github.com/Xinyuan-LilyGO/lilygo-micropython/blob/master/extmod/display/epd/README.md)

```python
import epd
import framebuf
epd47 = epd.EPD47()
epd47.power(True)
epd47.clear()
bitmap = bytearray(100*10*2)
fbuf = framebuf.FrameBuffer(bitmap, 100, 10, framebuf.GS4_HMSB)
fbuf.fill(0)
fbuf.text('MicroPython!', 0, 0, 0xffff)
fbuf.hline(0, 9, 96, 0xffff)
epd47.bitmap(list(bitmap),0,0,100,100)
```

But what is displayed is not correct.

scumola, 2022-03-01:
Ahh, I wasn't passing in the list of the one bytearray!  I am able to see the black blob that gets sent to the screen (not what's expected).  I tried different options for the framebuf.GS4_HMSB (all of the mono options too) and none of them seem to work as expected.  Is this a bug or am I just not using it properly?

scumola, 2022-03-01:
I think that if we could come up with good examples for working full-screen 1-bit framebuffer and 4-bit framebuffer, we could close this issue, but I've not been able to get that working yet.

scumola, 2022-03-01:
Also, every other time I try to get this to work, it reboots the board it seems

liangyingy, 2022-03-14:
> I think that if we could come up with good examples for working full-screen 1-bit framebuffer and 4-bit framebuffer, we could close this issue, but I've not been able to get that working yet.

```python
from math import floor

class FrameBuffer():
    def __init__(self, buffer, width, height):
        self.fb = buffer
        self.width = width
        self.height = height

    def fill(self, color):
        '''Fill the entire FrameBuffer with the specified color.
        '''
        for i in range(0, floor(self.width * self.height / 2)):
            self.fb[i] = color

    def pixel(self, x, y, color):
        '''Draw a pixel a given framebuffer.

        Args:
            x: Horizontal position in pixels.
            y: Vertical position in pixels.
            color: The gray value of the line (0-255).
        '''
        if x < 0 or x >= self.width:
            return
        if y < 0 or y >= self.height:
            return
        pos = y * floor(self.width / 2) + floor(x / 2)
        if floor(x % 2):
             self.fb[pos] = self.fb[pos] & 0x0F | color & 0xF0
        else:
            self.fb[pos] = self.fb[pos] & 0xF0 | color >> 4

    def hline(self, x, y, length, color):
        '''Draw a horizontal line to a given framebuffer.

        Args:
            x: Horizontal start position in pixels.
            y: Vertical start position in pixels.
            length: Length of the line in pixels.
            color: The gray value of the line (0-255);
        '''
        for i in range(0, length):
            self.pixel(x + i, y, color)

    def vline(self, x, y, length, color):
        '''Draw a horizontal line to a given framebuffer.

        Args:
            x: Horizontal start position in pixels.
            y: Vertical start position in pixels.
            length: Length of the line in pixels.
            color: The gray value of the line (0-255);
        '''
        for i in range(0, length):
            self.pixel(x, y + i, color)

    def circle(self, x, y, r, color):
        '''Draw a circle with given center and radius

        Args:
            x: Center-point x coordinate
            y: Center-point y coordinate
            r: Radius of the circle in pixels
            color: The gray value of the line (0-255)
        '''
        f = 1 - r
        ddF_x = 1
[... 256 lines cut ...]
if __name__ == "__main__":
    from epd import EPD47
    e = EPD47()
    e.power(True)
    e.clear()
    buffer = bytearray(480 * 540)
    fb = FrameBuffer(buffer, 960, 540)
    fb.fill(255)

    fb.hline(100, 100, 100, 256)

    fb.vline(100, 100, 100, 256)

    fb.circle(200, 200, 50, 256)
    fb.fill_circle(400, 200, 50, 256)

    fb.rect(600, 200, 100, 100, 256)
    fb.fill_rect(800, 200, 100, 100, 256)

    fb.line(100, 300, 400, 500, 256)
    fb.triangle(200, 400, 150, 500, 250, 400, 256)
    fb.fill_triangle(600, 400, 600, 500, 700, 500, 256)
    e.bitmap(list(buffer), 0, 0, 960, 540)

```

scumola, 2022-03-16:
Oh, wow!  Quite a bit of work.  I'm impressed!  I didn't know that you'd be re-writing all of the framebuffer functions yourself!  There isn't a way to just re-use all of the existing framebuffer libraries out there already - the ones that support fonts and everything?  Will I have to write my own font-rendering code in python in order to draw text to the framebuffer?

liangyingy, 2022-03-22:
support text https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/extmod/display/framebuf1

closed 2022-04-07 by liangyingy

skorokithakis, 2023-03-02:
I cannot get any of these examples to work, my screen keeps displaying white. I can only draw things if I change the offset on `bitmap()` to `20, 20`. The C++ demos work fine, though. Does anyone know what might be wrong?

kubark42, 2023-03-27:
@skorokithakis  I'm having a similar issue. Generally, the micropython EPD support seems to be troubled, and very little of the examples here work, and the few which do result in very poor quality images. When compared to the C++ examples, there's a stark difference. Perhaps something upstream changed and has caused regressions?