Xinyuan-LilyGO/lilygo-micropython #6: Build Failure

Martyn1066, 2022-04-01:
Hi, 
I would love to install this on my T5 EPD47, however I'm struggling to build the firmware. After several attempts where I've run out of space (at various stages), I think I've got everything necessary downloaded but the build fails with the following -

Checking Python dependencies...
Python requirements from /media/pi/128GB/lilygo-micropython/build_dir/esp32/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory /media/pi/128GB/lilygo-micropython/build_dir/esp32/LILYGO_EPD47
Executing "ninja all"...
ninja: error: build.ninja:18058: bad $-escape (literal $ must be written as $$)
ninja failed with exit code 1

any help would be greatly appreciated.

liangyingy, 2022-04-02:
1. You can try to delete the build_dir directory to see if the compilation is successful.
2. can you tell me what os are you using？
3. The attached firmware is compiled for me, you can use it directly

[LilyGo-MicroPython_esp32_LILYGO_EPD47_MPY-7c54b6428_IDF-e5f754b26c_B20220322_e9fa4d3f97.zip](https://github.com/Xinyuan-LilyGO/lilygo-micropython/files/8401550/LilyGo-MicroPython_esp32_LILYGO_EPD47_MPY-7c54b6428_IDF-e5f754b26c_B20220322_e9fa4d3f97.zip)

Martyn1066, 2022-04-02:
Hi,
Thanks for the reply, I'm using Raspberry Pi OS for desktop on an old netbook.
I've downloaded the firmware and will try it later, but it would be nice to be able to build it myself and mess with the functionality.
Should I be using VS Code to do the build?
Deleted the build_dir and trying a rebuild now.
Thanks

benner, 2022-04-03:
I have similar error during build:

```
-- Configuring done
-- Generating done
CMake Error:
  Running

   '/..../bin/ninja' '-C' '/..../lilygo-micropython/build_dir/esp32/LILYGO_EPD47' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:18064: bad $-escape (literal $ must be written as $$)

  

CMake Warning:
  Manually-specified variables were not used by the project:

    EXTMOD_FROZEN_DIR

```

OS: Ubuntu 21.10

liangyingy, 2022-04-03:
> I have similar error during build:
>
> ```
> -- Configuring done
> -- Generating done
> CMake Error:
>   Running
>
>    '/..../bin/ninja' '-C' '/..../lilygo-micropython/build_dir/esp32/LILYGO_EPD47' '-t' [...]

Do you have ninja installed on your computer?

benner, 2022-04-03:
yes, via `pip3 install ninja`:

```
% ninja --version
1.10.2.git.kitware.jobserver-1
```

liangyingy, 2022-04-03:
Please uninstall ninja and try to compile again

liangyingy, 2022-04-03:
I reproduced your problem. After installing ninja through `pip3 install ninja`, the compilation will go wrong. 

After uninstalling ninja, the compilation will be successful.

benner, 2022-04-03:
Removed `ninja` (including provided by distribution). Getting different error now:

```
[ 79%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/emitnthumb.c.obj
[ 80%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/emitnx64.c.obj
[ 80%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/emitnx86.c.obj
In file included from /....../lilygo-micropython/build_dir/micropython/py/builtin.h:29,
                 from /....../lilygo-micropython/build_dir/micropython/py/builtinhelp.c:30:
/....../lilygo-micropython/build_dir/micropython/py/builtinhelp.c: In function 'mp_help_print_obj':
/....../lilygo-micropython/build_dir/micropython/py/builtinhelp.c:131:32: error: 'MP_QSTR_modules' undeclared (first use in this function); did you mean 'MP_QSTR_module'?
     if (obj == MP_OBJ_NEW_QSTR(MP_QSTR_modules)) {
                                ^~~~~~~~~~~~~~~
/....../lilygo-micropython/build_dir/micropython/py/obj.h:96:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^~~
/....../lilygo-micropython/build_dir/micropython/py/builtinhelp.c:131:32: note: each undeclared identifier is reported only once for each function it appears in
     if (obj == MP_OBJ_NEW_QSTR(MP_QSTR_modules)) {
                                ^~~~~~~~~~~~~~~
/....../lilygo-micropython/build_dir/micropython/py/obj.h:96:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^~~
[ 80%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/emitnxtensa.c.obj
[ 80%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/emitnxtensawin.c.obj
make[4]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:725: esp-idf/main/CMakeFiles/__idf_main.dir/....../lilygo-micropython/build_dir/micropython/py/builtinhelp.c.obj] Error 1
...
```

liangyingy, 2022-04-03:
delete build_dir directory

benner, 2022-04-03:
Attaching full 
```
% bash -c "rm -fr lilygo-micropython && git clone https://github.com/Xinyuan-LilyGO/lilygo-micropython && cd lilygo-micropython && cp config_T5-4.7 config && make" > build 2>&1
```
[build.log](https://github.com/Xinyuan-LilyGO/lilygo-micropython/files/8404360/build.log).

liangyingy, 2022-04-03:
I suspect that the bug is caused by the use of multi-threaded compilation.

But I can't reproduce this bug.

Please try the method:

```shell
$ pwd
/home/lewin/lilygo-micropython
$ cd build_dir/esp32/esp-idf
$ git diff
diff --git a/tools/idf.py b/tools/idf.py
index 430f1ce0ba..31e9d111d0 100755
--- a/tools/idf.py
+++ b/tools/idf.py
@@ -74,7 +74,7 @@ GENERATORS = [
     ("Ninja", ["ninja"], ["ninja", "--version"], "-v"),
     (
         MAKE_GENERATOR,
-        [MAKE_CMD, "-j", str(multiprocessing.cpu_count() + 2)],
+        [MAKE_CMD, "-j", "1"],
         [MAKE_CMD, "--version"],
         "VERBOSE=1",
     ),
$ cd ../../../ && make
```

Martyn1066, 2022-04-03:
Hi, thanks liangyingy,

uninstalled ninja, deleted build_dir (again) and rebuilt. I haven't installed the  (my build) firmware yet, but its the same size as the one you (graciously) provided, which I have installed and playing with!
Again many thanks.
M

mamoit, 2022-04-24:
I banged my head against the wall with dependency issues too, therefore I implemented #7 to get a consistent build.
Hope this helps someone.

closed 2022-08-05 by liangyingy