Xinyuan-LilyGO/lilygo-micropython #12: Beginner guide ?

tonyrb, 2022-07-27:
Hello,
I am kinda loss, I have bought the 4.7 lilygo epaper board and trying to install micropython.

First question which version should I choose 4,7 Plus or 4,7 normal (i have the battery version)

So far I have built the image bin and now what should i do with it ?

Any guide to walk a newbie to help me integrate micropython ?

Thanks

liangyingy, 2022-07-27:
You should use 4,7 normal, its microprocessor is esp32, 4.7 plus's microprocessor is esp32s3.

For burning the image, you should refer to the esp32 burning method of micropython.

[image]

You can find your built image at this path:

lilygo-micropython\bin\esp32\LILYGO_T5-4.7

Or use our precompiled images:

https://github.com/Xinyuan-LilyGO/lilygo-micropython/releases/download/v0.1.0/LilyGo-MicroPython_esp32_LILYGO_EPD47_MPY-da4b38e75_IDF-e5f754b26c_B20220601_5263c3cfad.bin

This is the api documentation for epd.

https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/extmod/display/epd

This is framebuf1, which is specially tailored for 4.7

https://github.com/Xinyuan-LilyGO/lilygo-micropython/blob/master/extmod/display/framebuf1/framebuf1.py

tonyrb, 2022-07-27:
Amazing thank you for the quick answer !!

I manage to complete and install it, now fun begins ! 

I am on MacOs so for the next user I followed this guide using esptool instead of windows tool to update the firmware work like a charm.
https://pythonforundergradengineers.com/how-to-install-micropython-on-an-esp32.html

For the ssh serial monitor you can use screens instead of putty.

closed 2022-07-27 by tonyrb