Xinyuan-LilyGO/lilygo-micropython #13: LIB_JPEG: prepare error: Not supported JPEG standard when running standard example

tonyrb, 2022-07-29:
Hello,

I am trying to show a picture using the provided sample code and got this specific errors when trying to show an jpg image.

```
['boot.py', 'clean1.jpg', 'framebuf1.py']
>>> from epd import EPD47
>>> from random import randint
>>> from time import sleep
>>> 
>>> e = EPD47()
>>> 
>>> e.power(True)
>>> 
>>> e.clear()
>>> x = randint(0, 960-200)
>>> y = randint(0, 540-200)
>>> e.jpeg("/clean1.jpg", x, y)
**E (167478) LIB_JPEG: prepare error: Not supported JPEG standard**
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4252
load:0x40078000,len:12384
load:0x40080400,len:3392
entry 0x4008060c
W (58) boot: PRO CPU has been reset by WDT.
W (58) boot: WDT reset info: PRO CPU PC=0x400803c3
W (58) boot: WDT reset info: APP CPU PC=0x40082f7c
MicroPython 9293d59-dirty on 2022-06-01; ESP32 LILYGO EPD47 with ESP32
Type "help()" for more information.
```

Making the esp32 reboot automatically.

the file i am using and uploaded on the board is this one:
https://upload.wikimedia.org/wikipedia/commons/3/3f/JPEG_example_flower.jpg

Many thanks for the help !

liangyingy, 2022-08-01:
Please try another image

tonyrb, 2022-08-01:
You were right I downloaded another image reduced it using Pil library and now it's working for both jpg & jpeg version.

Sorry for this.

I have another issue with framebuf1 file (very slow to load and nothing on screen) I will open another issue.

closed 2022-08-01 by tonyrb