Xinyuan-LilyGO/lilygo-micropython #14: Provide example of the framebuf1 library ?

tonyrb, 2022-08-01:
Hello,

Many thanks for the previous support provided (really appreciate it).

I am trying to use the framebuf1 library proposed to draw some rectangles and draw line to build the user interface.

When running the code provided at the end of the library well the esp32 stay blocked (refresh two times but nothing to see) and need to remove usb to have access back to the repl.

here is the code:

```
  from FiraSansRegular10pt import FiraSansRegular10pt
  from framebuf1 import FrameBuffer

  buffer = [ 0 for _ in range(0, int(960 * 540 / 2)) ]
  fb = FrameBuffer(buffer, 960, 540)
  fb.fill(255)

  fb.hline(100, 100, 100, 256)

  fb.vline(100, 100, 100, 256)

  fb.circle(200, 200, 50, 256)
  fb.fill_circle(400, 200, 50, 256)

  fb.rect(600, 200, 100, 100, 256)
  fb.fill_rect(800, 200, 100, 100, 256)

  fb.line(100, 300, 400, 500, 256)
  fb.line(400, 500, 100, 300, 256)
  fb.line(400, 300, 100, 500, 256)
  fb.line(100, 500, 400, 300, 256)

  fb.triangle(200, 400, 150, 500, 250, 400, 256)
  fb.fill_triangle(600, 400, 600, 500, 700, 500, 256)

  fb.text(FiraSansRegular10pt, "1234", 50, 50)
  try:
      from epd import EPD47
      e = EPD47()
      e.power(True)
      e.clear()
      e.bitmap(buffer, 0, 0, 960, 540)
  except:
      print("The current parser is not micropython")
```

I have managed to run this code (previously shared on github issue)

```
import epd
import framebuf
epd47 = epd.EPD47()
epd47.power(True)
epd47.clear()
bitmap = bytearray(100*10*2)
fbuf = framebuf.FrameBuffer(bitmap, 100, 10, framebuf.GS4_HMSB)
fbuf.fill(0)
fbuf.text('MicroPython!', 0, 0, 0xffff)
fbuf.hline(0, 9, 96, 0xffff)
epd47.bitmap(list(bitmap),0,0,100,100)
```

But I only get a black box and that's pretty much it.

End goal is to draw the user interface (rectancle & line) maybe best to directly draw the ui from a jpeg and load it directly ? (and play with the local clear function to update anything) but I though this library would help me draw the interface.

If you can provide dumb example for drawing rectangle (hence i would know how to draw anything afterward) I will be much interested !

As always thank for the great support I received so far and apologize for the beginners question.

liangyingy, 2022-08-05:
Please try the following firmware:

[LilyGo-MicroPython_esp32_LILYGO_T5-4.7_MPY-da4b38e75_IDF-e5f754b26c_B20220805_cbd0167cbd.zip](https://github.com/Xinyuan-LilyGO/lilygo-micropython/files/9267713/LilyGo-MicroPython_esp32_LILYGO_T5-4.7_MPY-da4b38e75_IDF-e5f754b26c_B20220805_cbd0167cbd.zip)

Please refer to the following demo program:

[demo example](../tree/master/extmod/display/framebuf1/examples/demo)

Compatible with framebuf:

https://docs.micropython.org/en/v1.18/library/framebuf.html

closed 2022-08-19 by liangyingy

tonyrb, 2023-01-09:
Thank you ! (sorry for late answer it did help me a lot :)).