Xinyuan-LilyGO/lilygo-micropython #15: T5-4.7 EDP shows ghosted image

gddeen, 2022-09-21:
My E-Ink display shows the entire screen reprinted about 16 pixels more along the X axis.

The screen does occasionally show perfectly.

EXCELLENT work porting all of LilyGo ESP products to Micropython and Circuitpython.

My board was just purchased but there is some melted PLA material over the protective plastic on
the screen. Maybe the display has reached its end of life. Ive purchased several more units to check
if it is a hardware problem or software.

This is part of a system with a LoRa point to point sensor, transmitter, to LoRa receiver to MQTT broker.
This unit reads from the MQTT broker and displays the information from the LoRa sensors.

My code on the EDP47 display:
"""
import os
#__import__("maintest")
import machine
import time

#import sys
#sys.exit(0)

if __name__ == "__main__":
    ##from FiraSansRegular10pt import FiraSansRegular10pt
    from framebuf1 import FrameBuffer

    #buffer = bytearray(int(960 * 540 / 2))
    buffer = bytearray(int(190 * 540 / 2))
    fb = FrameBuffer(buffer, 190, 540)
    fb.fill(0)
    ##fb.hline(100, 100, 100, 255)

    ##fb.vline(100, 100, 100, 255)

    ##fb.circle(200, 200, 50, 255)
    ##fb.fill_circle(400, 200, 50, 255)
    ##for i in range(16):
    ##    fb.rect(0+i*17*2, 260, 33, 32, 255)
    ##    fb.fill_rect(0+i*2*17, 292, 33, 32, i*17)
    ##    fb.text(FiraSansRegular10pt, str(i*17),i*34+1, 280 )
    ##fb.rect(600, 200, 100, 100, 255)
    ###fb.fill_rect(800, 200, 100, 100, 255)

    ##fb.line(100, 300, 400, 500, 255)
    ##fb.line(400, 500, 100, 300, 255)
    ##fb.line(400, 300, 100, 500, 255)
    ##fb.line(100, 500, 400, 300, 255)

    ##fb.triangle(200, 400, 150, 500, 250, 400, 255)
    ##fb.fill_triangle(600, 400, 600, 500, 700, 500, 255)

    ##fb.text(FiraSansRegular10pt, "1234", 50, 50)
    ##fb.text(FiraSansRegular10pt,str(gc.mem_free()),50,66)
    #fb.rect(770, 1, 160, 538, 255)
    #fb.fill_rect(771, 2, 188, 537, 10)
    fb.rect(1, 1, 189, 538, 255)
    #fb.fill_rect(2, 2, 188, 537, 0)
    time.sleep(4)
    if True: #try:
        from epd import EPD47
        e = EPD47()
        e.power(True)
        #time.sleep(4)
        e.clear()
        #e.power(False)
        #time.sleep(4)
        #time.sleep(4)
        #e.bitmap(buffer, 0, 0, 960, 540)
        #EPD47.clear([x=0, y=0, width=960, height=540])
        #EPD47.power(active)
        #EPD47.bitmap(framebuffer, x, y, width, height)
        #EPD47.jpeg(pathname, x, y)
        #EPD47.text(text, x, y, font_size=12) # 9,12, 18, 24
        #EPD47.clear([x=0, y=0, width=960, height=540])
        #EPD47.width()
        #EPD47.height()
        e.jpeg("AtriaBackdrop.jpg",0,0)
        #time.sleep(2)
        y=32
        dataA = [
# TIME, Lat, Long, dhop, speed, Inclination, Direction, Temp, AQI, Rel Humidity, Odor, Batt1, Batt2            
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8),
([0,0,0,0,0,0,0],"35D56.000836' N","78D 55.50946' W", 3.8, 3678, (0, 0, 0), 100, "N", 73.2, 35, 63, "OK", 3.7, 23.8)
            ]
        data = dataA[0]
        e.text("Atria",         770+59,y*1,font_size=18)#9,12,18,24
        e.text("SouthPoint",    770   ,y*2,font_size=18)#9,12,18,24
        e.text("Walk",          770+63,y*3,font_size=18)
        e.text("Idle Seconds:", 770+3,y*5,font_size=9)
        e.text("Speed:",        770+3,y*6,font_size=9)
        e.text("Inclination:",  770+3,y*7,font_size=9)
        e.text("Direction:",    770+3,y*8,font_size=9)
        e.text("Temperature:",  770+3,y*9,font_size=9)
        e.text("AQI:",          770+3,y*10,font_size=9)
        e.text("Rel Humid:",    770+3,y*11,font_size=9)
        e.text("Fecal/Urine:",  770+3,y*12,font_size=9)
        e.text("Sensor Battery",770+3,y*13,font_size=9)
        e.text("Motor Battery", 770+3,y*14,font_size=9)
        txt=str(data[4])
        #print(txt,len(txt),range(len(txt)) )
        #print("A",txt[0],"B", txt[1],"C", txt[2],"D", txt[3])
        for i in range(0,len(txt)):
            #print(txt[i:1])
            e.text(str(txt[i]), 960-4-13*len(txt)+i*13, y*5 ,font_size=12)
        for i in range(0,len(str(data[5]))):
            e.text(str(data[5])[i], 960-4-13*len(str(data[5]))+i*13, y*6 ,font_size=12)
        for i in range(0,len(str(data[6]))):
            e.text(str(data[6])[i], 960-4-13*len(str(data[6]))+i*13, y*7 ,font_size=12)
        for i in range(0,len(str(data[7]))):
            e.text(str(data[7])[i], 960-4-13*len(str(data[7]))+i*13, y*8 ,font_size=12)
        for i in range(0,len(str(data[8]))):
            e.text(str(data[8])[i], 960-4-13*len(str(data[8]))+i*13, y*9 ,font_size=12)
        for i in range(0,len(str(data[9]))):
            e.text(str(data[9])[i], 960-4-13*len(str(data[9]))+i*13, y*10 ,font_size=12)
        for i in range(0,len(str(data[10]))):
            e.text(str(data[10])[i], 960-4-13*len(str(data[10]))+i*13, y*11 ,font_size=12)
        for i in range(0,len(str(data[11]))):
            e.text(str(data[11])[i], 960-4-13*len(str(data[11]))+i*13, y*12 ,font_size=12)
        for i in range(0,len(str(data[12]))):
            e.text(str(data[12])[i], 960-4-13*len(str(data[12]))+i*13, y*13 ,font_size=12)
        for i in range(0,len(str(data[13]))):
            e.text(str(data[13])[i], 960-4-13*len(str(data[13]))+i*13, y*14 ,font_size=12)
        #time.sleep(2)
        e.bitmap(buffer, 770, 0, 189, 540)
        #e.power(True)
        #time.sleep(8)
        e.power(False)
    #except:
    #    print("The current parser is not micropython")

"""

gddeen, 2022-09-21:
[image]
[image]

liangyingy, 2022-09-22:
The ghosted image problem, we have not yet found the cause.

#2 has been mentioned

sandzco, 2022-11-02:
Can you provide the epd.py or epd.mpy(just like you have provided the framebuf1.py file)? I am unable to get the EPD47() class from anywhere else.

johnongit, 2022-12-29:
Did you find the root cause ? 
I've same issue.

lbuque, 2022-12-30:
@sandzco 
> Can you provide the epd.py or epd.mpy(just like you have provided the framebuf1.py file)?

The epd.py or epd.mpy file cannot be provided, because the epd module needs to be implemented in c.

> Can the micropython firmware be provided for EPD47 just like the "Xinyuan-LilyGO/LilyGo-eink-v2.3-micropython" ?

You can get the latest firmware at the github action.
https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/runs/3552467553

johnongit, 2022-12-30:
Hello @lbuque 
What is the difference between T5 4.7 and T5 4.7 plus ?

lbuque, 2023-01-03:
@johnongit 
T5 4.7 plus uses esp32-s3 with better performance.
You can get details about the T5-4.7 Plus on [AliExpress](https://www.aliexpress.us/item/3256804461011991.html).

neilfrndes, 2023-03-21:
@lbuque is there an update for the ghosting issue?

hengying, 2023-04-03:
> You can get the latest firmware at the github action. https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/runs/3552467553

The link is expired, where can I download the latest firmware?

The firmware under this is old:
https://github.com/Xinyuan-LilyGO/lilygo-micropython/releases

adetauriac, 2023-05-02:
Hello,

  I just recieved T5 4.7 (not plus, error during my choice), I try to compile the framework, but I've issue with cmake.
I try the old fimeware : https://github.com/Xinyuan-LilyGO/lilygo-micropython/releases, same result with "gosted" on text 

`epd47.text("test 2", 30,30)`

Someone have better version of binary ? 

 Thanks

bkircher67, 2023-05-29:
Hello, I've finally managed to build the .bin-file for my T5-4.7 - but I see the same ghosting problem - but in contrary to to provided bin-file (https://github.com/Xinyuan-LilyGO/lilygo-micropython/releases/download/v0.1.0/LilyGo-MicroPython_esp32_LILYGO_EPD47_MPY-da4b38e75_IDF-e5f754b26c_B20220601_5263c3cfad.bin) the `e.bitmap(buffer ...` cmd works without crashing!! Are there any updates planed?