Xinyuan-LilyGO/lilygo-micropython #16: T-Display S3

hemna, 2022-09-30:
Any support for the LilyGo T-Display S3 ?

liangyingy, 2022-10-17:
micropython firmware for T-DisplayS3:
https://github.com/Xinyuan-LilyGO/lilygo-micropython/suites/8804879618/artifacts/400521365

The micropython source code of T-DisplayS3:
https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/target/esp32s3/boards/LILYGO_T-DisplayS3

Demo of lcd:
https://github.com/liangyingy/st7789_binding_micropython/tree/master/examples

hemna, 2022-10-17:
When running arduino IDE or esptool.py to try and upload any firmware or talk to the board I get a failure.

MacOS

```
└─> sudo esptool.py --port /dev/tty.usbmodem1234561 read_mac
Password:
esptool.py v4.3
Serial port /dev/tty.usbmodem1234561
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

I have micropython running on the board and can get into it with ampy commands

```
└─> ampy ls
/boot.py
/webrepl_cfg.py
```

```
└─> sudo esptool.py --chip esp32s3 --port /dev/tty.usbmodem1234561 write_flash -z 0 LilyGo-MicroPython_esp32s3_LILYGO_T-DisplayS3_MPY-d7919ea71_IDF-8153bfe412_B20221017_9950096a17.bin
esptool.py v4.3
Serial port /dev/tty.usbmodem1234561
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

hemna, 2022-10-17:
I have also tried from linux raspi

```
[ 1431.347720] usb 1-1.2: USB disconnect, device number 7
[ 1433.998372] usb 1-1.2: new full-speed USB device number 8 using dwc_otg
[ 1434.144261] usb 1-1.2: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 1434.144285] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1434.144301] usb 1-1.2: Product: Espressif Device
[ 1434.144316] usb 1-1.2: Manufacturer: Espressif Systems
[ 1434.144330] usb 1-1.2: SerialNumber: 123456
[ 1434.148880] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
```

```
(.venv) root@aprs-igate:/home/waboring/devel# esptool.py --port /dev/ttyACM0 read_mac
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

```
(.venv) root@aprs-igate:/home/waboring/devel# esptool.py --port /dev/ttyACM0 --trace read_mac
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.102 No serial data received.
```

russhughes, 2022-10-18:
Press and release the reset button while holding down the boot button. Release the boot button. Use esptool to flash then press the reset button once more.

closed 2022-10-20 by liangyingy