Xinyuan-LilyGO/lilygo-micropython #17: T-Display config doesn't build

hemna, 2022-10-01:
```
└─> cat config
CONFIG_TARGET=esp32
CONFIG_BOARD=LILYGO_T-Display
```

```
make[1]: Entering directory '/home/waboring/devel/micropython/lilygo-micropython/build_dir/esp32/micropython/ports/esp32'
idf.py -D MICROPY_BOARD=LILYGO_T-Display -B /home/waboring/devel/micropython/lilygo-micropython/build_dir/esp32/build-LILYGO_T-Display  -DUSER_C_MODULES=/home/waboring/devel/micropython/lilygo-micropython/extmod/micropython.cmake build
Executing action: all (aliases: build)
Running cmake in directory /home/waboring/devel/micropython/lilygo-micropython/build_dir/esp32/build-LILYGO_T-Display
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=LILYGO_T-Display -DUSER_C_MODULES=/home/waboring/devel/micropython/lilygo-micropython/extmod/micropython.cmake -DCCACHE_ENABLE=0 /home/waboring/devel/micropython/lilygo-micropython/build_dir/esp32/micropython/ports/esp32"...
CMake Error at CMakeLists.txt:18 (message):
  Invalid MICROPY_BOARD specified: LILYGO_T-Display

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
make[1]: *** [Makefile:34: all] Error 2
```

liangyingy, 2022-10-06:
Is the code up to date?

hemna, 2022-10-06:
```
└─> make clean
Clear working directory...
Clear working directory...ok
╭─lilygo-micropython on  master ❯
└─> git pull
Already up to date.
└─> make
Prepare working directory...
Prepare working directory...ok
Prepare micropython source code...
Push patch on /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/micropython
/bin/bash: quilt: command not found
/bin/bash: quilt: command not found
/Users/i530566/devel/mine/micropython/lilygo-micropython
Prepare micropython source code...ok
Prepare expansion module..

Requirement already satisfied: MarkupSafe>=2.0 in /Users/i530566/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Jinja2>=2.4->Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/requirements.txt (line 22)) (2.1.1)
All done! You can now run:

  . /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/export.sh

. /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/export.sh && make -C /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/micropython/ports/esp32 MICROPY_BOARD_DIR=/Users/i530566/devel/mine/micropython/lilygo-micropython/target/esp32/boards/LILYGO_T-Display BOARD=LILYGO_T-Display BUILD=/Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/build-LILYGO_T-Display USER_C_MODULES=/Users/i530566/devel/mine/micropython/lilygo-micropython/extmod/micropython.cmake EXTMOD_FROZEN_DIR=/Users/i530566/devel/mine/micropython/lilygo-micropython/extmod
Setting IDF_PATH to '/Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf'
Detecting the Python interpreter
Checking "python" ...
Python 3.8.7
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/i530566/.espressif/python_env/idf4.4_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/components/esptool_py/esptool
  /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/components/espcoredump
  /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/components/partition_table
  /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/components/app_update
  /Users/i530566/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin
  /Users/i530566/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin
  /Users/i530566/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin
  /Users/i530566/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin
  /Users/i530566/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /Users/i530566/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /Users/i530566/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
  /Users/i530566/.espressif/python_env/idf4.4_py3.8_env/bin
  /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

WARNING: Failed to load shell autocompletion for bash version: 3.2.57(1)-release!
idf.py -D MICROPY_BOARD=LILYGO_T-Display -B /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/build-LILYGO_T-Display  -DUSER_C_MODULES=/Users/i530566/devel/mine/micropython/lilygo-micropython/extmod/micropython.cmake build
Executing action: all (aliases: build)
Running cmake in directory /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/build-LILYGO_T-Display
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=LILYGO_T-Display -DUSER_C_MODULES=/Users/i530566/devel/mine/micropython/lilygo-micropython/extmod/micropython.cmake -DCCACHE_ENABLE=0 /Users/i530566/devel/mine/micropython/lilygo-micropython/build_dir/esp32/micropython/ports/esp32"...
CMake Error at CMakeLists.txt:18 (message):
  Invalid MICROPY_BOARD specified: LILYGO_T-Display

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
make[1]: *** [all] Error 2
make: *** [all] Error 2
...

```

liangyingy, 2022-10-07:
```shell
sudo apt-get install git make python3 python3-pip cmake quilt
```

closed 2022-10-07 by liangyingy