Xinyuan-LilyGO/lilygo-micropython #18: Build LilyGo T-PicoC3 error: Invalid MICROPY_BOARD specified: LILYGO_T_PicoC3

FreddyBI, 2022-10-17:
Hello,

T-PicoC3
$ cp config_T-PicoC3 config
$ sudo apt install gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential
$ make

How resolve this problem?

[image]

liangyingy, 2022-10-18:
```shell
sudo apt-get install git make python3 python3-pip cmake quilt
rm build_dir -rf
make
```

FreddyBI, 2022-10-18:
Thanks for help.

sudo apt-get install git make python3 python3-pip cmake quilt
rm build_dir -rf
make
-->  now other Build error
[image]
[image]
[image]

closed 2022-10-18 by FreddyBI

FreddyBI, 2022-10-20:
Hello,
I renstal all.
And now build is ok.
Thanks.