Xinyuan-LilyGO/lilygo-micropython #20: Build failure Invalid Micropy_board Lilygo_T-SIM7000G

rwslippey, 2022-10-27:
Just getting started with the board and trying to use micropython, getting a build failure though.
```
CMake Error at CMakeLists.txt:18 (message):
  Invalid MICROPY_BOARD specified: LILYGO_T-SIM7000G
```

liangyingy, 2022-10-28:
[image]

rwslippey, 2022-10-28:
> [image]

Will double check to make sure they installed but I did install the prerequisites per the documentation. Will report back later on today when I get home.

rwslippey, 2022-10-29:
Ok, an update here. Ive managed to get the build to work. Turns out a package failed to install first run and I missed it.

Next question, is there any documentation aside from this [old code example](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/MicroPython_LoBo), which isn't working currently with this build. Really, I'm just looking for where to start, network connection, gps, etc

besi, 2022-11-12:
I got the same error on a mac and all I need to get it to work was install quilt:

	brew install quilt

closed 2023-01-03 by lbuque