Xinyuan-LilyGO/lilygo-micropython #23: Build failure for T3-display-S3

Drew40B, 2022-12-04:
Hello,
I followed the directions for the T3-Display -S3 and got the below error. I am running Ubuntu 22.04 with python 3.8.2.

**Commands**
```
$ cp config_T-DisplayS3 config
$ make
```

**Output**
```
git config --global --add safe.directory /media/drew/DATA1/git/github/lilygo-micropython/tmp/dl/micropython/lib/wiznet5k
fatal: Failed to recurse into submodule path 'lib/asf4'
fatal: Failed to recurse into submodule path 'lib/axtls'
fatal: Failed to recurse into submodule path 'lib/berkeley-db-1.xx'
fatal: Failed to recurse into submodule path 'lib/btstack'
fatal: Failed to recurse into submodule path 'lib/cyw43-driver'
fatal: Failed to recurse into submodule path 'lib/fsp'
fatal: Failed to recurse into submodule path 'lib/libffi'
fatal: Failed to recurse into submodule path 'lib/libhydrogen'
fatal: Failed to recurse into submodule path 'lib/lwip'
fatal: Failed to recurse into submodule path 'lib/mbedtls'
fatal: Failed to recurse into submodule path 'lib/micropython-lib'
fatal: Failed to recurse into submodule path 'lib/mynewt-nimble'
fatal: Failed to recurse into submodule path 'lib/nrfx'
fatal: Failed to recurse into submodule path 'lib/nxp_driver'
fatal: Failed to recurse into submodule path 'lib/pico-sdk'
fatal: Failed to recurse into submodule path 'lib/stm32lib'
fatal: Failed to recurse into submodule path 'lib/tinyusb'
fatal: Failed to recurse into submodule path 'lib/wiznet5k'
unpack micropython-v1.19.tar.gz
tar: /media/drew/DATA1/git/github/lilygo-micropython/dl/micropython-v1.19.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
touch: cannot touch '/media/drew/DATA1/git/github/lilygo-micropython/build_dir/esp32s3/micropython/prereq': No such file or directory
make: *** [Makefile:42: micropython] Error 1
`
```

lbuque, 2022-12-05:
Try the following:

```shell
$ rm -rf build_dir
$ cp config_T-DisplayS3 config
$ make > build.log 2>&1
```

Drew40B, 2022-12-08:
Still failed.
[build.log](https://github.com/Xinyuan-LilyGO/lilygo-micropython/files/10189703/build.log)

Drew40B, 2022-12-08:
Thought: Is it not possible for someone to build and actually publish the result to https://www.micropython.org/download/?

lbuque, 2022-12-09:
> Still failed.
It should be a disk permission problem that caused the micropython source code download to fail.
You can try to mount the disk to the $HOME directory and give you the permissions of the current user.

> Thought: Is it not possible for someone to build and actually publish the result to https://www.micropython.org/download/?
You can get the latest firmware [here](https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/runs/3552467553).

Drew40B, 2022-12-24:
I tried a different tact I used this docker image mgos/esp32-build:4.4.1-r7, shelled into the container and ran the associated instructions. I got further but still failed. Attached is a copy of the latest log.

Can you please just build the firmware and make it publicly available somewhere? 

[build.log](https://github.com/Xinyuan-LilyGO/lilygo-micropython/files/10298782/build.log)

lbuque, 2023-01-03:
@Drew40B 
> Can you please just build the firmware and make it publicly available somewhere?

You can get the latest firmware [here](https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/runs/3552467553).

Drew40B, 2023-01-05:
@lbuque 
That link you posted only takes me to the build process. It does not take me to the final build destination

lbuque, 2023-01-06:
@Drew40B 
[image]

closed 2023-02-16 by lbuque