Xinyuan-LilyGO/lilygo-micropython #24: Prebuilt Firmwares?

NEXT-JP, 2022-12-20:
Are there any plans to include prebuilt firmwares for each of the supported LilyGo boards? Took me quite a while to build a firmware for the TSIM7000, and ultimately required offloading the task to my VPS, as it seems the M1 architecture is not supported [^1]. 

Prebuilt firmware would have saved a lot of time and headache...

[^1]: I think this was caused by an old version of gevent, which doesn't have a pyproject.toml, so it needed Python 3.8, which isn't really supported on M1. At least that was the best I could deduce.

lbuque, 2022-12-21:
There is precompiled firmware [here](https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/runs/3552467553)

NEXT-JP, 2022-12-21:
Cool, thanks for pointing those out. I see there are also firmware assets in the releases, though those are older.
I guess my follow up suggestion is add a quick link in the README to the most recent builds. 
Thanks again!

closed 2022-12-21 by NEXT-JP

kubark42, 2023-03-23:
@lbuque that link is dead. I'm guessing if I go to https://github.com/Xinyuan-LilyGO/lilygo-micropython/actions/, I can find some build artifacts and, using the left-hand nav bar, grab the one specific to my board?

BTW, I have a similar M1/M2 problem. I imagine there will be more and more people like @NEXT-JP and me. Any chance a new release could be made?

lbuque, 2023-03-25:
@kubark42 I created nightly-build ci. You can find precompiled firmware there.

https://github.com/lbuque/lilygo-micropython/actions/workflows/nightly-build.yml

[image]

ChrisKahrs, 2023-03-25:
I really want this to work and want to help in any way I can.  I tried the very generic S3 .bin from micropython site, the one from espress and he one here in the build artifacts...[LILYGO_T-DisplayS3_firmware](https://github.com/lbuque/lilygo-micropython/suites/10566969223/artifacts/525916122) but they all end up just pinging every second to my com port (in device manager) and if I open Thonny (it does that even when Thonny isn't open on my compuer) it just shows "invalid header" over and over.  Sorry, thanks for the help and getting a good .bin out for folks.  Hopefully once you have it, we can update the readme to reflect where they are.  Thanks!

kubark42, 2023-03-25:
> I really want this to work and want to help in any way I can. I tried the very generic S3 .bin from micropython site, the one from espress and he one here in the build [...]

@ChrisKahrs this is getting to some bad thread drift, since this thread is about firmware releases, but you can look at https://github.com/Xinyuan-LilyGO/lilygo-micropython/issues/28 to see how I flashed successfully. If that doesn't work for you, then you should probably open a new issue, including what you did and what you see.