Xinyuan-LilyGO/lilygo-micropython #27: A complete starter guide for MicroPython?

skipfire, 2023-01-29:
Is there a complete starter guide for using the LilyGo 4.7" with MicroPython?  I tried the precompiled firmware (found the link in a closed issue, https://github.com/Xinyuan-LilyGO/lilygo-micropython/releases/download/v0.1.0/LilyGo-MicroPython_esp32_LILYGO_EPD47_MPY-da4b38e75_IDF-e5f754b26c_B20220601_5263c3cfad.bin), and get "no module named 'epd'" when I tried the text-only code at https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/extmod/display/epd.  The firmware also doesn't have other basic MicroPython things like ureqeusts.

A simple starter guide for use with MicroPython with very small examples of how to get text to show and how to get images to show would be incredible.

lbuque, 2023-01-30:
You can try the latest [firmware](https://github.com/Xinyuan-LilyGO/lilygo-micropython/suites/9514697728/artifacts/451759539).

skipfire, 2023-01-30:
Ok, tried that and got some progress.  Added some print statements to the published sample code and I get to `cleared`, but no further.

```
from epd import EPD47

e = EPD47()
print("instance")
e.power(True)
print("powered")
e.clear()
print("cleared")
e.text("hello world", 100, 100, font_size=9)
e.text("hello world", 100, 120, font_size=12)
e.text("hello world", 100, 150, font_size=18)
e.text("hello world", 100, 190, font_size=24)
print("printed hello world")
del e
```

lbuque, 2023-02-03:
You can try to run the following code:

https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/extmod/display/framebuf1/examples/demo

Upload the two files inside to the board.

skipfire, 2023-02-03:
No good again, it clears but just hangs at `e.bitmap(buffer, 0, 0, 960, 540)`

lbuque, 2023-02-03:
[image]

lbuque, 2023-02-03:
@skipfire You can use this code to test whether the board is working properly: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/examples/demo

closed 2023-02-03 by skipfire

wszoecs, 2023-12-25:
For a T5.4 - I got the above example work:

* micropython-builder/extmod/display/framebuf1/examples/demo > mpremote cp * :

cp demo.py :
cp FiraSansRegular10pt.py :

* /micropython-builder/extmod/display/framebuf1/examples/demo > mpremote run demo.py

THIS definitely seems to WORK :-)