Xinyuan-LilyGO/lilygo-micropython #28: Document how to flash from the CLI

kubark42, 2023-03-25:
A couple users have asked for [beginners](https://github.com/Xinyuan-LilyGO/lilygo-micropython/issues/27) [guides](https://github.com/Xinyuan-LilyGO/lilygo-micropython/issues/12), so I thought it would be helpful to document in the `README.md` how to flash when using macOS or Linux.

Flashing from non-Windows systems is possible with the correctly formatted command. With `esptool`, the following worked for me:

```
"~/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool"  --chip esp32 --port /dev/tty.usbserial-54350321661 erase_flash
"~/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool" --baud 460800 --chip esp32 --port /dev/tty.usbserial-54350321661 write_flash -z 0x1000 ~/Downloads/LilyGo-MicroPython_esp32_LILYGO_T5-4.7_MPY-d7919ea71_IDF-8153bfe412_B20230222_a8ff3ecace.bin
```

Please note that to use this above command correctly on other computers, it is necessary to specify the locally correct serial port and binary blob names.

I also had to try a couple different baud rates for it to work 100% of the time. `460800` seems to be the fastest I can go and never experience uploading failures.

ChrisKahrs, 2023-03-25:
Ok, mine sees to work now, but only if I flash it to 0 not to 0x1000, odd?

kubark42, 2023-03-25:
> Ok, mine sees to work now, but only if I flash it to 0 not to 0x1000, odd?

Hah, that might solve my mystery! A while ago, I had flashed an SP 32S3 and I had take a note of the command I had use. The address was `0x0000`, and not `0x1000`. However, yesterday, when I was trying to flash my T5 –4.7 with that exact same command it would give me a similar error to the one you mentioned in https://github.com/Xinyuan-LilyGO/lilygo-micropython/issues/24#issuecomment-1483913505. Using `0x1000` resolved that problem. Maybe an ESP32-S3 needs `0x0000` and a regular ESP32 needs `0x1000`?

lbuque, 2023-05-06:
This is to keep consistent with the original micropython burning method.

brianwyld, 2023-05-08:
@ChrisKahrs Thanks for noting that the ESP32S3 firmware.bin MUST be flashed to 0x0000 NOT 0x1000 (like the ESP32 build).
I was completely stuck as to why none of the .bins for my new T5-PLUS units would run...

siddharth178, 2024-02-04:
Where can I find firmware for T3S3 v1.2 board? I don't see it listed in `Supported Boards` section of Readme. So I tried using the one available on micropython's website, but it fails too.

Has anyone successfully installed micropython on T3S3 v1.2?

Below command says that the firmware is not built for ESP32-S3, Can anyone help me with how to build it for S3?
```
esptool.py -b 115200 --port /dev/ttyACM0 write_flash -z 0x0000 LILYGO_TTGO_LORA32-20240105-v1.22.1.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: dc:54:75:e1:9c:44
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 9 but value was 0. Is this image for a different chip model?

A fatal error occurred: LILYGO_TTGO_LORA32-20240105-v1.22.1.bin is not an ESP32-S3 image. Use --force to flash anyway.
```