Xinyuan-LilyGO/lilygo-micropython #29: Micropython .uf2 or firmware.bin file?

ChrisKahrs, 2023-03-25:
I have this display and am very excited to use it.  I only know/use micropython so I was trying to find a .uf2 file or the like (.bin via the esp tool) to put on the device, but no luck.  It would seem that all windows computers would use the same bin (or at least most would use a standard one) so compiling one and posting on the site to get micropython users going without having to figure out c compilers and make files would open this up to a wider community.  Right now I just want to return it or give it 1 star as it is pretty useless to me.  Thanks!

lbuque, 2023-03-25:
@ChrisKahrs You can check this issue, which provides precompiled firmware.
#24

ChrisKahrs, 2023-03-25:
Ok, checking now.  Thanks!

ChrisKahrs, 2023-03-25:
Closing as this is covered in #30.  Thanks!

closed 2023-03-25 by ChrisKahrs