Xinyuan-LilyGO/lilygo-micropython #31: Examples for LCD on T-Display S3?

clydebarrow, 2023-03-29:
I've built the Micropython binary and flashed it to a T-Display S3. I can run simple programs, but have no idea where to start to drive the LCD. I see there is an `lcd` module available, but have no idea how to use it.

Are there any examples for this (or something equivalent?)

inlanger, 2023-04-15:
Have the same issue here. The `lcd` module can't be used to display data, just for basic screen tests. You can find an example of "usage" here - https://github.com/Xinyuan-LilyGO/lilygo-micropython/blob/master/target/esp32s3/boards/LILYGO_T-DisplayS3/modules/inisetup.py

It looks like we need to use `st7789`, which should be included in the firmware, but it's not for some reason, or I can't find it.

jetbalsa, 2023-05-30:
Digging into this more, russhughes/st7789_mpy does not get compiled it, it gets downloaded, but never included in the build process. only the LCD module is. There are examples of using it with a framebuffer, Check out the ` build_dir/esp32s3/lcd_binding_micropython/examples` for some examples of using the LCD lib

lbuque, 2023-05-30:
@jetbalsa Currently russhughes/st7789_mpy is only used on T-PicoC3.