Xinyuan-LilyGO/lilygo-micropython #38: Compilation error: fatal: reference is not a tree: 59bbbdae5ecc9ae84b986ce54c089250a13cb768

cescp, 2023-07-29:
Hi,
I'm trying to compile the project with:
```
cp config_T-DisplayS3 config
make
```
but I get the following error:
```
fatal: reference is not a tree: 59bbbdae5ecc9ae84b986ce54c089250a13cb768
```
If I change the content of the file extmod/display/lcd/Makefile so as it points to version 0.1.0 instead of commit 59bbbdae5ecc9ae84b986ce54c089250a13cb768 (which it seems it does not exist):
```
PKG_SOURCE_VERSION:=0.1.0
```
then compilation can proceed.

lbuque, 2023-07-29:
@cescp This is the repo I maintain which is ahead of lilygo-micropython. You can try it.

https://github.com/lbuque/micropython-builder

wszoecs, 2023-12-25:
indeed - THAT version does work and produces the required flash-bins.
But what I discovered is:
* centos-8 stream: build works fine.
* fedora-39: build fails with cython related errors. :-( - haven't found a way to workaround it.